ROBOTIS-GIT/open_manipulator #239: Open Manipulator cannot launch - process dies with exit code -11

RubenCarreira99, 2022-09-29:
Hi,

I have a waffle Pi with an open manipulator attached to it. I connected the OpenCR to the raspberry PI that controls the robot and installed ROS2 foxy distro on it. Now, I'm following the documentation in order to be able to control the open manipulator but when I insert the launch command, it fails with the error below.

[image]

Has anyone ever came across this?

RubenCarreira99, 2022-09-30:
Before launching open_manipulator, I connected the OpenCR to my laptop and used arduinoIDE to flash the usb_to_dxl example. Then I installed the DynamixelWizard2 to scan the servos IDs and, lastly, I connected the OpenCR to my raspberry again and used the launch command. It worked for me.

yun-goon, 2024-12-19:
@RubenCarreira99 Apologies for the delayed response. We have assembled a TF team to address OpenManipulator-x issues and are actively working on resolutions. Please note that we do not support EOL (End-of-Life) ROS versions; our support currently extends to Noetic and Humble versions. We kindly ask that you try again with the latest version. If the same issue arises, please feel free to resubmit it, and we will address it promptly.

closed 2024-12-19 by yun-goon