ROBOTIS-GIT/open_manipulator #240: ROS2 Humble package installation - colcon fails to build packge

ajaysonar, 2022-10-05:
Hello,

I am trying to setup Openmanipulator-X (Ubuntu 22.04 + ROS2 Humble). While following the instruction on,
https://emanual.robotis.com/docs/en/platform/openmanipulator_x/quick_start_guide/#install-ubuntu-on-pc

after executing, colcon build --symlink-install, I get the following error...

Starting >>> open_manipulator_x_controller
--- stderr: open_manipulator_x_controller                                
/home/ros2desk/colcon_ws/src/open_manipulator/open_manipulator_x_controller/src/open_manipulator_x_controller.cpp: In member function ‘void open_manipulator_x_controller::OpenManipulatorXController::init_parameters()’:
/home/ros2desk/colcon_ws/src/open_manipulator/open_manipulator_x_controller/src/open_manipulator_x_controller.cpp:66:26: error: no matching function for call to ‘open_manipulator_x_controller::OpenManipulatorXController::declare_parameter(const char [4])’
   66 |   this->declare_parameter("sim");
      |   ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~

AND...

/home/ros2desk/colcon_ws/src/open_manipulator/open_manipulator_x_controller/src/open_manipulator_x_controller.cpp:67:26: error: no matching function for call to ‘open_manipulator_x_controller::OpenManipulatorXController::declare_parameter(const char [15])’
   67 |   this->declare_parameter("control_period");
      |   ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~

All I could tell was it has something to do with the "sim" and "control_period" parameters. Any feedback would be appreciated.

Thanks!

irenebm, 2022-11-16:
The parameters should be declare with default values. Thus, that code section would be:

`this->declare_parameter("sim", default);
  this->declare_parameter("control_period", default);`

Take into account that the parameter type is inferred from the default value.

For example, in this case could be:

`this->declare_parameter("sim", false);
  this->declare_parameter("control_period", 0.010);`

So _sim_ would be set to a bool type and _control_period_ would be set to a double type, which are the types assing in the yaml file.

Aki1608, 2023-06-21:
@irenebm, Thanks this solves the issue.

Gredin67, 2023-07-12:
@irenebm @Aki1608 @ajaysonar Can someone open a pull request to solve this issue ?

I tried to replace
```
  // Declare parameters that may be set on this node
  this->declare_parameter("sim");
  this->declare_parameter("control_period");
```
with
```
this->declare_parameter("sim", default);
this->declare_parameter("control_period", default);
```
and I'm still getting 
```
error: expected primary-expression before ‘default’
   66 |   this->declare_parameter("sim", default);

```
Should I change something in the parameter file ? `src/open_manipulator/open_manipulator_x_controller/param/open_manipulator_x_controller_params.yaml `

```
open_manipulator_x_controller:
  ros__parameters:
    sim: false
    control_period: 0.010
```

ROBOTIS-Will, 2023-07-12:
Thanks for reviewing this issue.
We're working on OpenMANIPULATOR-X package for Humble and Iron.
Based on current progress, I'm expecting to update the code by mid August at the latest, but you may see some WIP branches soon.
I'll make sure to reflect this information in the update.
Thanks!

yun-goon, 2024-12-16:
Hi
The newly released [Humble package](https://github.com/ROBOTIS-GIT/open_manipulator/tree/humble) has resolved the simulation issues. Please use this package for your setup. For detailed information and installation instructions, please refer to the [e-Manual](https://emanual.robotis.com/docs/en/platform/openmanipulator_x/quick_start_guide/).

sunghowoo, 2025-01-13:
@ajaysonar This issue will be closed due to prolonged inactivity. Please try again with the Humble version, and if the issue persists, feel free to reopen it.

closed 2025-01-13 by sunghowoo