ROBOTIS-GIT/open_manipulator #242: default tele-operations freezing

gbud, 2022-10-28:
Hello,

I am having some issues running the default keyboard tele-operations.  I am able to enter several inputs in the tele-operation terminal but am not allowed enter any more at some point; this terminal will remain on the previous command.  Additionally, the initial terminal that launched the manipulator shows several error that have to do with missing libraries?  I downloaded the packages in the instructions when installing ROS and am not sure what other ones I may be missing.  Attached is a screen shot of the terminals for reference.

Thank you,
Greg

[image]

ChangAnCN, 2024-04-23:
same issue

yun-goon, 2024-12-05:
Hi.
This issue has been resolved in the `develop` branch. The manipulator is now set to ignore new inputs while it is in motion. 
Thank you!

sunghowoo, 2025-01-13:
Refer to Yun-goon's answer for initial guidance. This issue will be closed due to prolonged inactivity. If the issue persists, feel free to reopen it. We recommend using the Humble version. Thank you!

closed 2025-01-13 by sunghowoo