ROBOTIS-GIT/open_manipulator #243: chaining together multiple tasks

arturzarske, 2022-11-17:
Hello, 
I am working with ROS Noetic under Ubuntu 20.04. I would like to chain together multipe tasks for the robot to go along a grid. I am currently operating the robot with the normal GUI. Is there a convinient way to put in several positions in the task space for the robot to go along and wait a few seconds at each position before goint to the next? I hope I made clear what I am trying to accomplish. I hope there is someone that an help me.

Thanks to everyone in advance!

yun-goon, 2024-12-16:
Hi @arturzarske 
The functionality you are looking for is not implemented in the ROS Noetic package for the OpenManipulator. However, this feature has been added in the ROS2 Humble package. The Humble version supports chaining multiple tasks for the robot to move along a grid or predefined points, including the ability to add pauses between positions.

If possible, I recommend upgrading to ROS2 Humble and using the updated package for your application. You can find the documentation and setup instructions for the Humble version .
[Manual](https://emanual.robotis.com/docs/en/platform/openmanipulator_x/overview/)
[Humble package](https://github.com/ROBOTIS-GIT/open_manipulator/tree/humble)

Feel free to ask if you need further assistance!

sunghowoo, 2025-01-13:
@arturzarske Please refer to yun-goon's answer for initial guidance. This issue will be closed due to prolonged inactivity.

closed 2025-01-13 by sunghowoo