ROBOTIS-GIT/open_manipulator #244: MoveIt failed to open the port

arturzarske, 2022-11-22:
Hey,

I am currently working with ROS Noetic and Ubuntu 20.04. and have worked my way the through the manual. At step 5.4.1.2. I encounter the error shown in the screenshot. I don't know what to do next. Maybe someone here can help me. Thanks in advance!! 
[image]

yun-goon, 2024-12-16:
Hi @arturzarske 
The error indicates that the communication port could not be opened. When using the OpenManipulator for the first time, it is necessary to set up the communication interface. Please follow the steps provided in the [Communication Interface section of the OpenManipulator-X manual](https://emanual.robotis.com/docs/en/platform/openmanipulator_x/quick_start_guide/#communication-interface).

Once the communication interface is properly configured, try running the launch file again. If the problem persists, verify the USB port and permissions as outlined in the manual.

Thank you.

sunghowoo, 2025-01-13:
@arturzarske Please refer to yun-goon's answer for initial guidance. This issue will be closed due to prolonged inactivity. if the issue persists, feel free to reopen it.

closed 2025-01-13 by sunghowoo