ROBOTIS-GIT/open_manipulator #247: Roll implementation in Task space for OpenManipulator-X

AdityaPrakash23, 2023-03-04:
Hello there!
I am working on a project where I need to implement roll angle for the gripper for an inspection operation using ROS Noetic. My team has already made the URDF model, made the changes in Gazebo yaml files, and other changes such as adding the extra joint in the openmanipulator.cpp file and other changes mentioned in the Issue [84](https://github.com/ROBOTIS-GIT/open_manipulator/issues/84). I am using the keyboard teleop, which also has been modified, to control the robot in simulation.  
When moving the roll joint in joint space, it works perfectly and the roll angle movement shows up even in Gazebo. But when I try to move the new roll joint in the Task space, the joint shows no motion at all. When I try to check the ROS topic that is supposed to have the current joint angle, that remains frozen as well, despite it working for movement in joint space commands. I also tried implementing pitch movement in task space since its not originally implemented, but that work perfectly fine.  
I have tried unsuccessfully to debug this error and my only suspect is the Kinematics. Could anyone give an expert opinion on this matter?  
Thanks!

shiba-8ro, 2023-03-05:
@AdityaPrakash23 
ROBOTIS Japan has released a model of Turtlebot3 equipped with 6DOF type OM-X.
It might be useful for you.

https://github.com/ROBOTIS-JAPAN-GIT/tb3mm_6DoF

AdityaPrakash23, 2023-03-06:
@shiba-8ro thank you for this. The thing is I am stuck using the OpenManipulator-X for my project and need the Roll implementation to work. I am suspecting the Kinematics.cpp might be needing some modification despite me defining the joint chain properly.

AdityaPrakash23, 2023-03-07:
@shiba-8ro thanks for sharing that! I still need something specific to my problem telling me what are the exact steps for modification and how to adjust kinematics for an extra joint, if needed.

shiba-8ro, 2023-03-07:
ROBOTIS Japan will create a document for adding extra joints to OM-X mainly for Robocup@Home players, but it still takes time to complete it.
It's a pity that we cant help you at this time.

AdityaPrakash23, 2023-03-07:
I see. But could share some pointers from your knowledge of the code? I modified the URDF files, yaml files and teleop_keyboard code. I also configured the main manipulator file to adjust the robot chain. My joint space movement for the new joint is working is working perfectly. Its only the task space movement that is not working. I have looked through the code to check the program flow with breakpoints. I assume the kinematics code should adjust for the joint chain adjustments in the manipulator code, but I feel there might be some adjustments required there.

yun-goon, 2024-12-03:
Hi @AdityaPrakash23 
The OpenMANIPULATOR-X gripper is designed as a 4-DOF system, which does not support roll angle rotation. Additionally, the current implementation relies on custom kinematics, and extending the number of axes (such as adding a roll joint) has not been considered in its design.

If roll functionality is required, significant modifications to both the hardware and kinematics framework would be necessary. Unfortunately, this is beyond the scope of the current OpenMANIPULATOR-X design.
Thank you.

closed 2025-01-13 by sunghowoo