ROBOTIS-GIT/open_manipulator #252: ros2_controller for ROS2 humble

Aki1608, 2023-06-22:
Hello everyone,

This is more like a question than an issue. I have the actual hardware and wanted to know if there is any ros2_controller plugin for the open manipulator arm so that I can control the arm using rviz?

I can move the arm using teleop but I cant control it with Rviz. I think I have to change the hardware plugin in the ros2_controller.xacro file. But I don't have any hardware plugin for open_manipulator_x. I found a similar one, but it is for "turtlebot3_open_manipulator" and not for manipulator_arm alone. 

Can anyone help me or guide me in finding/creating the plugin? Or any similar plugin is available? 
I am using ROS2 humble on Ubuntu 22.04.

Thank you.

yun-goon, 2024-12-16:
Hi @Aki1608 
The newly released ROS 2 [Humble package](https://github.com/ROBOTIS-GIT/open_manipulator/tree/humble) for the OpenManipulator-X is compatible with ROS 2 Control and also includes integration with MoveIt, allowing you to control the arm directly in RViz. We recommend using this package for better compatibility and ease of use.

Additionally, you can utilize the [Dynamixel Hardware Interface](https://github.com/ROBOTIS-GIT/dynamixel_hardware_interface), which is specifically designed to provide ROS 2 Control compatibility for Dynamixel-based hardware. This package can help you configure your hardware plugin for the OpenManipulator-X arm.

sunghowoo, 2025-01-13:
Please refer to yun-goon's answer for initial guidance. This issue will be closed due to prolonged inactivity.

closed 2025-01-13 by sunghowoo