ROBOTIS-GIT/open_manipulator #254: Package 'open_manipulator_x_controller' not found

Leo5017, 2023-09-05:
Package 'open_manipulator_x_controller' not found: "package 'open_manipulator_x_controller' not found, searching: ['/home/ubuntu/turtlebot3_ws/install/turtlebot3', '/home/ubuntu/turtlebot3_ws/install/turtlebot3_teleop', '/home/ubuntu/turtlebot3_ws/install/turtlebot3_bringup', '/home/ubuntu/turtlebot3_ws/install/turtlebot3_node', '/home/ubuntu/turtlebot3_ws/install/turtlebot3_example', '/home/ubuntu/turtlebot3_ws/install/turtlebot3_description', '/home/ubuntu/turtlebot3_ws/install/ld08_driver', '/opt/ros/foxy']"

Leo5017, 2023-09-05:
When I write this following command:
ros2 launch open_manipulator_x_controller open_manipulator_x_controller.launch.py usb_port:=/dev/ttyACM0
it the error has happened:
Package 'open_manipulator_x_controller' not found: "package 'open_manipulator_x_controller' not found, searching: ['/home/ubuntu/turtlebot3_ws/install/turtlebot3', '/home/ubuntu/turtlebot3_ws/install/turtlebot3_teleop', '/home/ubuntu/turtlebot3_ws/install/turtlebot3_bringup', '/home/ubuntu/turtlebot3_ws/install/turtlebot3_node', '/home/ubuntu/turtlebot3_ws/install/turtlebot3_example', '/home/ubuntu/turtlebot3_ws/install/turtlebot3_description', '/home/ubuntu/turtlebot3_ws/install/ld08_driver', '/opt/ros/foxy']"

yun-goon, 2024-12-16:
To resolve the issue, follow these steps:

Build and source the workspace:

```bash
colcon build
source install/setup.bash
```
Check if the package is built successfully:

```bash
ros2 pkg list | grep open_manipulator_x_controller
```
Run the launch file:

```bash
ros2 launch open_manipulator_x_controller open_manipulator_x_controller.launch.py usb_port:=/dev/ttyACM0
```
If the package is still not found, ensure it exists in your source directory and rebuild the workspace.

sunghowoo, 2025-01-13:
@Leo5017 Apologies for the delayed response. We have assembled a TF team to address OM-X issues and are actively working on resolutions. This issue will be closed due to prolonged inactivity.  if the issue persists, feel free to reopen it.

closed 2025-01-13 by sunghowoo