ROBOTIS-GIT/open_manipulator #255: Unable to initialize the OpenManipulator-x

Leo5017, 2023-09-06:
Running Ubuntu 20.04 on vmware
ROS2 Foxy

Hi I am a totally beginner of dynamixels and ros2
I have a couple of XM430-W210 for Turtlebot and five motors of XM430-W350, trying to build my own robot arm with the OpenCR
I have installed the ros2 foxy and the OpenCR but failed to start the controller:

$ ros2 launch open_manipulator_x_controller open_manipulator_x_controller.launch.py usb_port:=/dev/ttyACM0
[INFO] [launch]: All log files can be found below /home/leo5017/.ros/log/2023-09-05-19-57-11-076641-ubuntu-14683
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [open_manipulator_x_controller-1]: process started with pid [14685]
[ERROR] [open_manipulator_x_controller-1]: process has died [pid 14685, exit code -11, cmd '/home/leo5017/colcon_ws/install/open_manipulator_x_controller/lib/open_manipulator_x_controller/open_manipulator_x_controller /dev/ttyACM0 1000000 --ros-args -r __node:=open_manipulator_x_controller --params-file /home/leo5017/colcon_ws/install/open_manipulator_x_controller/share/open_manipulator_x_controller/param/open_manipulator_x_controller_params.yaml'].

I try to connect to the servos with the Dynmixel Wizard 2.0, they work fine.
and I also tested with the read_write_node example, they also work fine.

yun-goon, 2024-12-13:
Hi,
Thank you for reaching out. Please note that we do not officially support running on a virtual machine like VMware, as it can sometimes cause unexpected issues with hardware communication. We recommend running ROS2 Foxy and the OpenManipulator-X software directly on a native Ubuntu installation instead of using a VM. This approach should help eliminate potential compatibility or performance issues caused by virtualization.

If you encounter further problems after switching to a native setup, feel free to provide additional details, and we'll assist you further!

sunghowoo, 2025-01-13:
@Leo5017 Apologies for the delayed response. Please refer to yun-goon's answer for initial guidance. This issue will be closed due to prolonged inactivity.

closed 2025-01-13 by sunghowoo