ROBOTIS-GIT/open_manipulator #256: teleop issue in humble-devel and ros2-devel, ros2.

Eunsunggcu, 2024-03-10:
I want to control open manipulator 6dof in ros2 humble.
so I got the source code of open manipulator 4dof (branch:humble-devel) and tried to change it.

and i realize that teleop in humble-devel branch, there is nothing about gripper operation. 
[image]

but i see that part of teleop in foxy-devel.
[image]

Is that any reason about this difference between those two branches?

basalte1199, 2024-04-23:
It will work if you add the program related to gripper as shown in foxy-devel. Note that there is no send_tool_controll_request function.

yun-goon, 2024-12-16:
Hi, @Eunsunggcu 
In the newly released [humble package](https://github.com/ROBOTIS-GIT/open_manipulator/tree/humble), gripper control is available for operation. However, please note that this package is specifically designed for a 4DOF manipulator. Since you're using a 6DOF version, you'll need to modify the code accordingly to support the additional degrees of freedom. If you need assistance with adapting the code, feel free to ask!

sunghowoo, 2025-01-13:
Please refer to yun-goon's answer for initial guidance. This issue will be closed due to prolonged inactivity. If the problem persists, feel free to reopen it, and we will address it promptly. Thank you for your understanding!

closed 2025-01-13 by sunghowoo