ROBOTIS-GIT/open_manipulator #257: using teleop and stoppd responding

kaloflower, 2024-04-03:
just got teleop to work, although the inverse kinematics weren't working, the joints and preset locations were working, so i was excited. I hit w again to see if i got the same error in the controller terminal (not teleop terminal), and the teleop terminal stopped responding. I exitted out of everything and tried to open up the controller again, but got an error saying "stack smashing detected" and that all processes on the machine had died. Not sure how to proceed as I have absolutely no idea what went wrong. A side note about why the inverse kinematics weren't working - I believe that it is because we have 3 xl430s, and i was getting an error when i launched the controller - failed to set current based position control model. 
Any help is appreciated

ChangAnCN, 2024-04-23:
same issue

yun-goon, 2024-12-16:
Hi,

The error "failed to set current-based position control mode" indicates that your setup does not support this control mode. The OpenManipulator-X requires compatible motors (like the XM series) to properly execute inverse kinematics and current-based position control. Since you're using XL430s, which do not support this mode, this could be the root cause of the issues.

Additionally, the keyboard teleoperation issue has been resolved in the latest [Humble package](https://github.com/ROBOTIS-GIT/open_manipulator/tree/humble). We recommend using the updated Humble package for better stability and functionality. You can refer to the [OpenManipulator-X eManual](https://emanual.robotis.com/docs/en/platform/openmanipulator_x/quick_start_guide/) for detailed instructions.

Let us know if you need further assistance!

Thank you.

sunghowoo, 2025-01-13:
@kaloflower Please refer to yun-goon's answer for initial guidance. This issue will be closed due to prolonged inactivity. If the problem persists, feel free to reopen it, and we will address it promptly. Thank you for your understanding!

closed 2025-01-13 by sunghowoo