ROBOTIS-GIT/open_manipulator #258: open manipulator-x with ros2 simulation

ssapssabi, 2024-05-01:
Hi. i am trying to simulate openmanipulator-x with ros2 on gazebo, but i can't find any manual about simulate openmanipulator-x with ros2. So i'm wondering that how can i simulate openmanipulator-x with ros2.
sorry for my english skill. please let me know the method

zeidk, 2024-05-01:
Have you looked here?
https://emanual.robotis.com/docs/en/platform/turtlebot3/manipulation/#manipulation

sunghowoo, 2024-09-27:
@ssapssabi Sorry for the late response. Our team is currently expanding development resources for the open platform, and we are planning to update the Open Manipulator-X in November to make it compatible with Moveit 2 in ROS 2.

sunghowoo, 2024-12-13:
@ssapssabi The Humble MoveIt2 support version for OpenManipulator-X has been released.
You can check it at the following links:

[Quick Start Guide](https://emanual.robotis.com/docs/en/platform/openmanipulator_x/quick_start_guide/)
[GitHub Repository](https://github.com/ROBOTIS-GIT/open_manipulator/tree/humble)

closed 2024-12-13 by sunghowoo