ROBOTIS-GIT/open_manipulator #267: unable to use the gui interface

zainib716, 2025-02-15:
unable to use the gui interface, it shutdown when I selected joint space
roslaunch open_manipulator_control_gui open_manipulator_control_gui.launch

sunghowoo, 2025-03-04:
@zainib716 Would you be interested in trying ROS 2 Humble? The GUI version in this release is quite stable. Let me know if you need any assistance with the transition! https://emanual.robotis.com/docs/en/platform/openmanipulator_x/ros_operation/#gui-program

zainib716, 2025-03-04:
Dear Sungho
Thanks for your reply, actually my research required the ROS noetic version for open manipulator x simulation using moveit in rviz. I am struggling to plan the path and control the collision avoidance stuff as the gui interface is not added properly. I am a beginner in using ROS, not sure about creating the gui interface. Need your advice in this regard. May thanks

yun-goon, 2025-03-10:
Hello @zainib716 !

We didn’t use MoveIt in the Noetic version, but in the Humble version, MoveIt is fully integrated, and the GUI is also supported.

If you don’t necessarily need to use Noetic, I recommend analyzing and utilizing our package in the Humble version, as it provides a more stable and feature-complete experience.

Feel free to let me know if you encounter any issues while using it!

yun-goon, 2025-03-18:
This issue will be closed due to prolonged inactivity. If the problem persists, feel free to reopen it, and we will address it promptly. Thank you for your understanding!

zainib716, 2025-03-18:
Thanks for getting back. Actually my PhD project demands Noetic as recommended by my supervisor because most of the work we are discussing is made on noetic.
Should I discuss with my supervisor to change to humble? What do you suggest.

Regards
Zainib Bilal

zainib716, 2025-03-19:
Hi Yun
I have soted with the cmake and rqt, please can you guide me about computer
vision integration for this simulation. Many Thanks

yun-goon, 2025-03-20:
Hello @zainib716 

Noetic will reach End of Life (EOL) in May, so I personally recommend migrating to Humble if possible. Since most of your work is based on Noetic, it would be best to discuss this with your supervisor.

If you must proceed with Noetic, you will need to develop the MoveIt integration yourself using our source code as a reference. We can assist with bug fixes and questions related to our code, but MoveIt for Noetic will require custom implementation on your end.

I recommend checking out the MoveIt documentation to help with implementation:
👉 [MoveIt Tutorials](https://moveit.github.io/moveit_tutorials/)

Let me know if you need further assistance!