ROBOTIS-GIT/open_manipulator #291: Error in simulate openmanipulatorx hardware using moveit

MohdZaheen123, 2025-05-01:
I followed the steps mentioned in the manual to use openmanipulatorx hardware using moveit. when i plan and execute the trajectory using move it it is showing the below error in the terminal

`[ INFO] [1746117279.242346199]: ParallelPlan::solve(): Solution found by one or more threads in 0.011665 seconds
[ INFO] [1746117279.244137083]: SimpleSetup: Path simplification took 0.001652 seconds and changed from 3 to 2 states
[ INFO] [1746117279.245464049]: Returned 0 controllers in list
[ERROR] [1746117279.245545184]: Unable to identify any set of controllers that can actuate the specified joints: [ joint1 joint2 joint3 joint4 ]
[ERROR] [1746117279.245592009]: Known controllers and their joints:

[ERROR] [1746117279.245650600]: Apparently trajectory initialization failed
[ INFO] [1746117279.263244942]: ABORTED: CONTROL_FAILED

`

Woojin-Crive, 2025-06-19:
Hi @MohdZaheen123 !

We are currently preparing a major source code update and doing our best to get it merged as soon as possible.
We will notify you immediately once the update is completed!

Woojin-Crive, 2025-06-19:
The major update 4.0.0 has been released! Please check our new codebase.

Woojin-Crive, 2025-06-25:
Case closed due to no further comments received.