ROBOTIS-GIT/open_manipulator #297: Can't control OpenManupilator-y gripper with keyboard teleoperation in Gazebo Simulation

HieuBui99, 2025-06-04:
I'm trying to use the keyboard teleoperation function in Gazebo with OpenManupilator-y

Commands used in 2 separate terminals inside a docker container 
```bash
ros2 launch open_manipulator_bringup gazebo.launch.py
ros2 run open_manipulator_teleop keyboard_control_y
```
I can control the other 6 joints without issue but can't seem to control the gripper for some reason.
Terminal log when pressing o/p:
```
INFO] [1749027629.765405061] [keyboard_controller]: Sending gripper command: 0.10000000000000596
[INFO] [1749027629.766188835] [keyboard_controller]: Arm command sent: [-1.876138928184356e-09, -3.000961923152411e-07, 9.740005808017587e-07, -1.2553221358596965e-06, -7.437574605010626e-08, 2.41386584909248e-06]
[INFO] [1749027629.902235333] [keyboard_controller]: Sending gripper command: 0.20000000000000595
[INFO] [1749027629.902974265] [keyboard_controller]: Arm command sent: [-1.876138928184356e-09, -3.000961923152411e-07, 9.740005808017587e-07, -1.2553221358596965e-06, -7.437574605010626e-08, 2.41386584909248e-06]
[INFO] [1749027630.061389874] [keyboard_controller]: Sending gripper command: 0.30000000000000593
[INFO] [1749027630.062775718] [keyboard_controller]: Arm command sent: [-1.876138928184356e-09, -3.000961923152411e-07, 9.740005808017587e-07, -1.2553221358596965e-06, -7.437574605010626e-08, 2.41386584909248e-06]
[INFO] [1749027630.560867450] [keyboard_controller]: Sending gripper command: 0.4000000000000059
[INFO] [1749027630.561658253] [keyboard_controller]: Arm command sent: [-1.876138928184356e-09, -3.000961923152411e-07, 9.740005808017587e-07, -1.2553221358596965e-06, -7.437574605010626e-08, 2.41386584909248e-06]
```

I also tried with OM-X and was able to control its gripper.

yun-goon, 2025-06-13:
Hello!

Please open your .bashrc file using the `vi ~/.bashrc` command. Then, add `export ROBOT_MODEL=om_y` to the very end of the file.

After that, reload it with `source ~/.bashrc`, and then run Gazebo.

Thank you.

HieuBui99, 2025-06-13:
Hi!
Thank you for your answer. 
I have already modified the Dockerfile to have `export ROBOT_MODEL=om_y` by default when I ran the above commands. I also verified the variable value with `echo`. 
But it still doesn't work

HieuBui99, 2025-06-24:
Thank you! Everything is working correctly now.